ninjablocks/433Utils #12: Data received multiple times

ksb86, 2015-12-29:
I'm getting the data transfer ok by using 'sudo ./codesend 123' on one raspberry pi and 'sudo ./RFSniffer' on another raspberry pi.  However, for every time I transmit by using codesend, the sniffer end is receiving anywhere from 1 to 5 responses f
[image]
[image]

or each send.  Am I missing something? Is this how rf is supposed to work?  Thanks.

maditnerd, 2015-12-29:
Since 433mhz radio communication are unreliable, the code are sent multiples times to be sure it is received.

Here is where it is setup in RCSwitch library
https://github.com/ninjablocks/433Utils/blob/master/RPi_utils/RCSwitch.cpp#L41
````
this->setRepeatTransmit(10);
````

ksb86, 2016-01-02:
@maditnerd Thanks! That make sense.

rotv, 2016-02-10:
I'm guessing this issue can be closed since this is the intended behavior.

ksb86, 2016-02-11:
Yes, I would say so. Thanks.