ninjablocks/433Utils #21: RFSniffer no data

Thijxx, 2016-04-04:
I have an Arduino sending data over 433MHz, this is wokring fine and I receive the data on my Pi using the VirtualWire lib and PIGPIO. But when I run RFSniffer it shows no data. The PINs are the same so that can't be it. Any advice on how to find the problem?

Connor-Knabe, 2016-04-07:
I'm having the same issue I cannot seem to read anything from my receiver (I've tried two different receivers on different pins).

barrytuenter, 2016-04-17:
Same here :-(
I installed a new sd card several times, and thit exactly what it need to do but i can't recieve any signals.
I also try'd with another 433 receiver.

rotv, 2016-05-04:
Have you ruled out pulse length as a cause (are you sure you're running RFSniffer with the correct pulse length)?

Connor-Knabe, 2016-05-06:
No I wasn't aware that was something I could try. How do I go about that @rotv?

rotv, 2016-05-07:
@Connor-Knabe, I did that for my remotes by using [PISCOPE](http://abyz.co.uk/rpi/pigpio/piscope.html) where I could see the signals and measure the pulses.

In order to test different pulse lengths you can invoke RFSniffer with the optional parameter:
Usage: ./RFSniffer [<pulseLength>]
You can for instance try out different "50-step" values: 150, 200, 250...400

E.g.
sudo ./RFSniffer 150
_(for 150 microsecond/us pulses)._

My system, FHT-7901, uses 150 us pulses but the 433-Utils/rc-switch default setting of 350 also seems to work when I send codes. I'm using a better version of 433Utils which contains the latest and greatest rc-switch (git clone -b rcswitch --recursive git://github.com/rotv/433Utils.git). I've been trying to get it merged but no one is responding for the ninjablocks repositories.

floler, 2016-06-01:
+1
My RPi setup worked fine using _codesend_ and _RFSniffer_. I updated my rpi this morning, now running:
`Linux raspi 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux`

Right after the update _RFSniffer_ stopped working. I checked the reciever module on my other RPi (not updated, yet) where it's still working fine.

bartounet16, 2016-07-08:
Hello I don't understand
I have installed Fresh raspian 
and compile RFSNIffer with default Options ( int PIN = 2;)
I have connect rasp as this picture: http://www.homautomation.org/wp-content/uploads/2013/09/RPi.jpg
But i don't receive some data !!!
is there a problem with raspberry pie 2 and 3 ??

roidayan, 2016-07-16:
hi @bartounet16 , when I sniff data on the pi I need to touch(or almost touch) the remote to the receiver to actually make it catch the signal. trying even few cm away doesn't catch the signal at all. have you tried that?

667bdrm, 2016-08-03:
The problem reproducing on the latest Raspbian on B+ too. Tried the XY-MK-5V and RXB6 receivers, changed power supply and receiver power isolation. The both receivers works perfect on Arduino and ESP8266. As I could remember the XY-MK-5V has been worked on the older raspbian before the implementing kernel device tree.

dubanoze, 2016-08-03:
Guys, 
GPIO by default PULL DOWN.
When i add code for PULL UP sniffer start work.

floler, 2016-08-03:
@dubanoze feel free to start a pull request with a working version :smile:

dubanoze, 2016-08-03:
Wii paste code tomorrow. 
RPI on work.

rcoduk, 2016-08-12:
@dubanoze Are u still gonna paste that code :)

yat0tv, 2016-08-13:
Yeah I was wondering as well, I have some untested receivers that I can't get to spit out any data and hoping it's a pi/software issue and not hardware.

dubanoze, 2016-08-14:
``
root@raspberrypi:/home/pi/433/433Utils# git diff .
diff --git a/RPi_utils/RFSniffer.cpp b/RPi_utils/RFSniffer.cpp
index 7532429..11a71f3 100644
--- a/RPi_utils/RFSniffer.cpp
+++ b/RPi_utils/RFSniffer.cpp
@@ -22,13 +22,16 @@ int main(int argc, char *argv[]) {
      // This pin is not the first pin on the RPi GPIO header!
      // Consult https://projects.drogon.net/raspberry-pi/wiringpi/pins/
      // for more information.
-     int PIN = 2;
+     int PIN = 0;

      if(wiringPiSetup() == -1) {
        printf("wiringPiSetup failed, exiting...");
        return 0;
      }

+
+    pullUpDnControl(PIN, PUD_OFF);
+
      int pulseLength = 0;
      if (argv[1] != NULL) pulseLength = atoi(argv[1]);

diff --git a/rc-switch b/rc-switch
index c564517..62b9ebf 160000
--- a/rc-switch
+++ b/rc-switch
@@ -1 +1 @@
-Subproject commit c5645170be8cb3044f4a8ca8565bfd2d221ba182
+Subproject commit 62b9ebfc40667daf8a383c2e598dddbf89f297ff-dirty
``

dubanoze, 2016-08-14:
need add 
    pullUpDnControl(PIN, PUD_OFF);  

to RFSniffer.cpp

dubanoze, 2016-08-14:
root@raspberrypi:/home/pi/433/433Utils/RPi_utils# ./RFSniffer

Received 3224434
Received 3224434
Received 3224434
Received 3224436
Received 3224436
Received 3224436
Received 3224436

yat0tv, 2016-08-14:
hmm I've added that to line 38 and still not seeing any results.

dubanoze, 2016-08-14:
Can you connect oscilloscope to line ?

yat0tv, 2016-08-14:
I do not have that kind of hardware.

rotv, 2016-08-14:
You can use [PISCOPE](http://abyz.co.uk/rpi/pigpio/piscope.html).

girl71, 2016-08-16:
@rotv Hi, I use ./RFSniffer before and I receive 5592512 ,..........now I dont get any thing ,why??
I install  PISCOPE now what??

yat0tv, 2016-08-17:
I can see it on channel 27 of PISCOPE modulating up and down quite rapidly but nothing shows up for RFSniffer

electronicsguy, 2016-11-23:
It's working perfectly for me, on my single Raspberry Pi B+. Running Wheezy (Linux raspberrypi 4.1.19+) and no need of adding the pull up/down line mentioned above. 

All I did was start RFSniffer in one terminal and use codesend <string> from another terminal.

`Received 54321`
`Received 54321`

SamDecrock, 2017-05-05:
I had the same problem with one of two 433 Mhz receivers. One worked on the Pi, the other didn't. However, both worked on the Arduino!

Putting pin 2 into the "no pull up/down"-state fixed it like @dubanoze said.
You can do this from command line using

```
gpio mode 2 tri
```

I've tested this on both the Raspberry Pi Model B+ and Raspberry Pi 3.

The 100% usage is always there as I suppose WiringPi "polls" the pin every x microseconds.

Also, don't forget that the WiringPi PIN 2 is actually the 13th pin on the board, see: https://pinout.xyz/pinout/pin13_gpio27

Thijxx, 2017-05-05:
This is no longer an issue for me because I moved to ESP8266, I also use RadioHead, the successor of VirtualWire at the moment.

closed 2017-05-05 by Thijxx