ninjablocks/433Utils #23: RFSniffer symbol lookup error (ubuntu xenial)

lachuzhnikov, 2016-04-22:
./RFSniffer: symbol lookup error: /usr/local/lib/libwiringPi.so: undefined symbol: pthread_create

afeierman, 2017-07-04:
I'm also having this issue, running Ubuntu MATE 16.04. Is this a persistent issue, or there a workaround?

mirzamushic, 2017-08-17:
did anyone find solution?

afeierman, 2017-08-22:
@mirzamushic I ended up using Tim Leland's (very) similar solution: https://github.com/timleland/rfoutlet

JRamos29, 2018-11-19:
I'm also using ubuntu mate 16.04 and solved this issue changing the symlink in /usr/local/lib/libwiringPi.so (before pointing to libwiringPi.so.2.46) to /usr/lib/libwiringPi.so.2.32. as follows: 

lrwxrwxrwx 1 root root     22 Nov 17 23:05 libwiringPiDev.so -> libwiringPiDev.so.2.46
-rwxr-xr-x 1   root root  28900 Nov 17 23:05 libwiringPiDev.so.2.46
lrwxrwxrwx 1 root root     28 Nov 18 06:00 libwiringPi.so -> /usr/lib/libwiringPi.so.2.32