ninjablocks/433Utils #27: Compile error?

Dragao75, 2016-07-19:
Hi all, I am having trouble compiling. I get errors like 
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_join'
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_create'
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/libwiringPi.so: undefined reference to `pthread_cancel'
collect2: error: ld returned 1 exit status

I have read that using the latest gcc version compiler require that libraries follow the object or source files.

I am using ubuntu 16.04 and the latest gcc.. 

Anyone have any suggestion(s) on how to proceed?

Dragao75, 2016-08-11:
Everyone on vacation?

mat1989, 2016-08-12:
Hi all, I am also not able to compiling. I get this error:
make: *** No rule to make target '../rc-switch/RCSwitch.o', needed by 'send'.  Schluss.

Could someone help me please?

rotv, 2016-08-12:
Are you using a previously checked out git clone? After a recent PR the instructions have changed, the recursive part is new:

> git clone --recursive git://github.com/ninjablocks/433Utils.git (recursive ensure that the rc-switch submodule gets populated which is needed by RPi_utils)

If that doesn't do it, then something must have gone wrong with PR #17. I'll try to take a look at it during the weekend, meanwhile you can try https://github.com/rotv/433Utils

rotv, 2016-08-12:
I just verified that the latest 433Utils does indeed compile, so that leads me to believe that the 'recursive' was missing for @mat1989. Unfortunately I don't know whats wrong for @Dragao75, seems to be something outside of the 433Utils suite.

Dragao75, 2016-08-13:
@rotv 
I have to modify the Makefile to get compilation working.
I changed: CXXFLAGS=-DRPI   ->>> CXXFLAGS=-DRPI -pthread 
Had to add -pthread

closed 2016-08-13 by Dragao75

ChewieB, 2016-12-22:
That worked great for me too. I was following the [this](http://tutorials-raspberrypi.com/let-raspberry-pis-communicate-with-each-other-per-433mhz-wireless-signals/) tutorial and hit a brick wall. Thanks to you and Google :+)

OSevangelist, 2018-05-20:
i had to compile by adding -lrt -lcrypt to the compile flags