ninjablocks/433Utils #40: Unable to open GPIO edge interface

timherrm, 2017-07-12:
Hi,

i'm running Armbian 5.31 stable on an Banana Pi Pro.

If i try ./RFSniffer i get the following error:

`gpio: Unable to open GPIO edge interface for pin 27: Permission denied`

But i am already logged in as root. 

Could you please help me?

robototechnic, 2019-02-06:
Same problem. It seems that the problem is in the libwiringPi library. If manually entered
```
root@bananapi:~# gpio edge 27 in
gpio: Unable to open GPIO edge interface for pin 27: Permission denied
```
displays the same error

robototechnic, 2019-02-06:
After starting the ./RFSniffer in the directory `/sys/class/gpio/`appears
>lrwxrwxrwx 1 root root    0 Feb  6 14:35 gpio27 -> ../../devices/platform/soc@01c00000/1c20800.pinctrl/gpiochip0/gpio/gpio27

but it should be `gpio274` [because](http://wiki.banana-pi.org/Getting_Started_with_M2P#GPIO_Control). Our pin is called [PI18](http://wiki.lemaker.org/BananaPro/Pi:Pin_definition#Banana_Pi_CON3_definition)
Same problem [here](http://forum.lemaker.org/thread-22425-1-1.html)