ninjablocks/433Utils #42: rc-switch error

5dollaralien, 2017-08-17:
Hi,
I am very new to this sort of thing but am learning very quickly..well maybe
trying to compile the 433Utils and wiringPi on a Rpi 2 I get this error

pi@raspberrypi:~/433Utils/RPi_utils$ make all
g++ -DRPI   -c -o codesend.o codesend.cpp
codesend.cpp:14:22: fatal error: RCSwitch.h: No such file or directory
 #include "RCSwitch.h"
                      ^
compilation terminated.
<builtin>: recipe for target 'codesend.o' failed
make: *** [codesend.o] Error 1

but yet RCSwitch.h is in the rc-switch folder, is there something I am missing, believe me being new to linux have been missing quite a bit. Any help would be greatly appreciated

underwoodblog, 2017-08-18:
maby problems with access/file permissions? wrong user?

5dollaralien, 2017-08-18:
Thanks for reply underwoodblog,
I have not changed from the default user "pi" probably a bad idea now that we are talking about it. Been using that for everything. But for the permissions, could anyone tell me what that file would need, I know it does not have execute, but does have read(anyone) write(user only) if my memory serves me. I am willing to try anything
Thank You

hmueller01, 2017-08-18:
Have you cloned with `git clone --recursive git://github.com/ninjablocks/433Utils.git` to fetch rc-switch?

You can check this if you find a "rc-switch" subdir with "RCSwitch.h" and others in your "~/433Utils" directory.

5dollaralien, 2017-08-18:
Yes I did install using recursive, the rc-switch folder is there and so is the RCSwitch.h file which has text and code inside when I look at it on my pi. Here is what I found looking at the folders and the files through the terminal

pi@raspberrypi:~$ cd /home/pi/433Utils
pi@raspberrypi:~/433Utils$ ls -la
total 44
drwxr-xr-x  7 pi pi 4096 Aug 13 12:23 .
drwxr-xr-x 22 pi pi 4096 Aug 17 22:37 ..
drwxr-xr-x  6 pi pi 4096 Aug 13 12:23 Arduino_sketches
drwxr-xr-x  2 pi pi 4096 Aug 13 12:23 CHIP_utils
drwxr-xr-x  9 pi pi 4096 Aug 13 12:23 .git
-rw-r--r--  1 pi pi  228 Aug 13 12:23 .gitignore
-rw-r--r--  1 pi pi   88 Aug 13 12:23 .gitmodules
-rw-r--r--  1 pi pi 1088 Aug 13 12:23 LICENSE
drwxr-xr-x  3 pi pi 4096 Aug 13 13:47 rc-switch
-rw-r--r--  1 pi pi 2434 Aug 13 12:23 README.md
drwxr-xr-x  3 pi pi 4096 Aug 17 21:47 RPi_utils
pi@raspberrypi:~/433Utils$ cd
pi@raspberrypi:~$ cd /home/pi/433Utils/rc-switch
pi@raspberrypi:~/433Utils/rc-switch$ ls -la
total 76
drwxr-xr-x  3 pi   pi    4096 Aug 13 13:47 .
drwxr-xr-x  7 pi   pi    4096 Aug 13 12:23 ..
drwxr-xr-x 11 pi   pi    4096 Aug 13 12:23 examples
-rw-r--r--  1 pi   pi      34 Aug 13 12:23 .git
-rw-r--r--  1 pi   pi     150 Aug 13 12:23 .gitignore
-rw-r--r--  1 pi   pi    1137 Aug 13 12:23 keywords.txt
-rw-r--r--  1 pi   pi   20953 Aug 13 12:23 RCSwitch.cpp
-rw-r--r--  1 pi   pi    4983 Aug 13 12:23 RCSwitch.h
-rw-r--r--  1 root root 12060 Aug 13 13:47 RCSwitch.o
-rw-r--r--  1 pi   pi    1228 Aug 13 12:23 README.md
-rw-r--r--  1 pi   pi     744 Aug 13 12:23 .travis.yml
pi@raspberrypi:~/433Utils/rc-switch$ ls -la

the files are there but for some reason beyond me it says it is not

hmueller01, 2017-08-19:
This is strange, as latest codesend.cpp includes
`#include "../rc-switch/RCSwitch.h"`
in line 17. But your error occurs on line 14 and the #include does not include "../rc-switch/". Do you really have the latest version? It looks like the version from Mar 14, 2013 ...

I can only recommend to clone a new/latest version of the 433Utils.

5dollaralien, 2017-08-19:
I am so sorry I see now hmueller01, I looked at some modified files that I was instructed to put into the 433Utils/RPI_utils and just what you said, line 14 #include "RCSwitch.h" 
it was further back in the instructions when I first began an alarm system and forgot until you mentioned it. I think I need this guys code to run the alarm but need of course things to run correctly from the start. First off how do I go about getting back to original, of course I did not back up those two files. Can I just replace with winscp? I am sorry forgot to mention the modification that would have in retrospect cleared things up.

5dollaralien, 2017-08-19:
Thank You hmueller01 for your tip on the rc-switch call. apparently in the newly written code they were only calling for RCSwitch.h...after overthinking it for a while and rereading what you said it finally hit me just change it in the code, so far it is working good and pulling in codes. No errors finally!!