ninjablocks/433Utils #43: Legth of ./codesend

TheWesleyMan, 2017-09-09:
Hi,
I want to a particularly set of numbers to my ' power socket switch device'
but it sent something other then what i asked for.

I’ll explain it:

I’ve two terminals open one for sending and one for monitoring what is send (with RFSniffer)

if it sent this:
5 2 4 6 9 9 7
i see exactly that pair of numbers in RFSniffer

But if is send this:
4 1 0 0 1 4 3 0
I come's out like this: 
7 4 4 6 9 9 8
and as you see that totally different and one character shorter.

but if i send 
4 1 0 0 1 4 3
i see exactly the same as what is sended

5246997 (working pair)
41001430 (Pair with sending problems)
4100143 (one character shorter, and its working)

So if I send 7 characters it's good but if go over that it does something weird.

has this problem something to do with the maximum length of characters?
Or has it something to do with my adapter?
I hope you can help me out, because my home automation project is still on this moment.

i can try many things for you if its needed.

Martin-Laclaustra, 2017-09-24:
You are sending an integer longer than 24 bits (the default bit-length sent... see line 54 in codesend.cpp).
```
41001430 = 10 01110001 10100001 11010110
7446998 =     01110001 10100001 11010110
4100143 =     00111110 10010000 00101111
```
Longer number will not be received either unless you change parameters (or even type of variables) in rcswitch.

TheWesleyMan, 2017-09-24:
So if I change line 54 in codesend.cpp  it should be fine?

Martin-Laclaustra, 2017-09-25:
Please try it and report back where the hurdles are.
You may get in trouble in the following items:
- Transmitting: size of the variable that holds the code in rcswitch (32 bits).
- Receiving: size of the buffer that waits for the codes, which is specified as max-changes in a macro definition in rcswitch.
Good luck.

SiberaIndustries, 2022-12-02:
Check this fix #81 or check it out [here](https://github.com/SiberaIndustries/433Utils). Let me know if it solves your issue. Cheers

Also related to https://github.com/ninjablocks/433Utils/issues/66 or https://github.com/ninjablocks/433Utils/issues/80 or #63