ninjablocks/433Utils #52: Unable to import 433Utils into IDE Library

lrv209, 2018-03-27:
Hello,
I have downloaded the 433Utils zip file but when attempting to import into the Arduino IDE I get the following error

Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Arduino/Genuino Uno"

Specified folder/zip file does not contain a valid library

Hope someone can advise
Regards

per1234, 2018-03-27:
That's because it's not a library. From https://github.com/ninjablocks/433Utils/blob/master/README.md:
>It consists of two main sections- Arduino sketches and Raspberry Pi command line utilities.

Just unzip the downloaded file, then use the Arduino IDE's File > Open menu to open any of the sketches in the Arduino_sketches subfolder. Note the other part of the [Installation Instructions](https://github.com/ninjablocks/433Utils/blob/master/README.md#installation) you evidently didn't read:
>Install any required libraries:
Arduino [rc_switch ](http://code.google.com/p/rc-switch/) library

That library you will be able to install via the Arduino IDE's **Sketch > Include Library > Add .ZIP Library...**

lrv209, 2018-03-27:
Thanks for the response, I fortunately had realised that was the problem after deciphering the readme.md file but thanks anyway and also thanks for your admonition on what I read or didn't read. Always good to get advice from those who know so much more than us newcomers as it makes us feel so welcome. 
Regards

per1234, 2018-03-27:
You're quite welcome.

In the future when you solve the problem you reported in an issue please take the time to close the issue so others don't waste time providing assistance you no longer need.