ninjablocks/433Utils #63: Negative numbers due to higher-bit signals

maresb, 2018-12-24:
While running `RPi_utils/RFSniffer` I get negative numbers from my remote:

    Received -1550055680

If I instead use `pigpio` as per the instructions on [sending rf codes with pigpio](https://raspberrypi.stackexchange.com/a/78899/96477), I see that it is a 32-bit code which was not being correctly read:
code=2325481216 bits=32

I can transmit the correct code if below the line `tx=_433.tx(pi, gpio=TX)` I add the line:

    tx.set_bits(32)

kiki51, 2019-10-18:
I maresb. 
 
I have the same issue using my remote and the rfsniffer command. 

Have you find a solution ?
The code sent by the codesend command is good or bad?

Thanks in advance

maresb, 2019-10-22:
@kiki51 I don't remember whether or not codesend works properly with 32-bit codes.  I recommend switching to pigpio if that is possible for you.

SiberaIndustries, 2022-12-02:
I had the same issue and fixed it in https://github.com/ninjablocks/433Utils/pull/81 or check it out [here](https://github.com/SiberaIndustries/433Utils)

Also related to #66 or #80