ninjablocks/433Utils #66: Issue sending negative code

kiki51, 2019-10-18:
Hi 

I use rpiutils tools to send and receive code over 433mhz with my remote/plug. 

I use it under home assistant to power on /off my plug. It replace my remote. 

I have buy another set of remote and plug ( otio model. Ref 09ha2) which works fine. 
But when I use rfsniffer command to find the remote code sent by the remote, the result is a negative code like : -1094193664. 

After that I use codesend with this code and the sniffer return : 
Received 13102592. 

So I can’t send a negative code, it’s translate like a positive code. And it doesn’t work. 

Can you help me to solve this trouble?
Thanks. 
Sorry for my bad English. 
Eric

Henykau94, 2020-03-04:
I have the same problem can someone please help

SiberaIndustries, 2022-12-02:
The reason for the negative values is called "[integer overflow](https://en.wikipedia.org/wiki/Integer_overflow)". Your transmitter sends a code higher than the current data type (signed integer) can store, its like you want to count to 6 with 1 hand. You can count to 5 and then you have to start again for 5 to 6. Here its a signed int [-2147483648, +2147483647], just a bigger number.

Fixed it in #81 or check it out [here](https://github.com/SiberaIndustries/433Utils)