ninjablocks/433Utils #78: 433MHz module not working with RPI4

pijoe56, 2022-03-21:
Hi,
I have been using these modules with my Rasperry Pi3 and Zero and it has always worked perfectly, now I have tried it with the Raspberry Pi4 and despite the fact that the codes are sent it does not work. can someone help me, thanks.

[code]RF_COMMAND='/home/pi/433Utils/RPi_utils/codesend'
remote2off = '123456'
subprocess.call([RF_COMMAND, remote2off]) # remote2 switch off
...[/code]

lxne, 2023-10-28:
I have a new 433MHz module and I can't get it to work with my Pi4, too.
Any ideas or does it even work for anyone?

lxne, 2023-10-28:
With [rpi-rf](https://pypi.org/project/rpi-rf/) receiving works.
But my sender still isn't working. Maybe it is defect.